riscv/riscv-debug-spec #10: Can 'busy' be set without 'halted'?

mwachs5, 2017-02-05:
With the current encoding, if debugger writes a command that has prehalt and postexec bits set, there is no way to distinguish between "hart hasn't halted yet" and "hart already did the command". Or is it meant that 'busy' would be set immediately when the command is written, even before the hart halts? I guess that makes sense, since the *command* is executing. But I want to clarify and then have to update my diagram to match.

timsifive, 2017-02-05:
My intention was that the command execution part of the DM is busy. As soon
as a command is written, busy gets set, and it isn't cleared until the
command is done. I'll add that to the spec.

mwachs5, 2017-02-06:
Fixed by f511b056

closed 2017-02-06 by mwachs5