riscv/riscv-debug-spec #23: "Serial_select" description bug

zhutmost, 2017-02-15:
First thank you for working on it.

In ver 0.12, Serial_port_select is part of <dmcontrol>, but in 0.13, it is moved to <serial>.

The description in 7.11.17 is not correct now.

[image]

timsifive, 2017-02-15:
I don't understand what exactly is wrong with the current description of serdata. Can you explain in a bit more detail?

mwachs5, 2017-02-15:
There is no serial field in dmcontrol anymore. It's in serstatus.

mwachs5, 2017-02-15:
And serstatus should probably be "serialcs" to match the fact that it holds control and status and all other such registers are called "cs"

mwachs5, 2017-02-15:
I'll make a  PR for this

zhutmost, 2017-02-16:
Thanks. 
Markdown cannot display angle brackets. Actually I typed `part of <dmcontrol>` but the contents in the angle brackets aren't displayed. 
I'm sorry I don't make it clear.

zhutmost, 2017-02-16:
And which branch is the main one? The master branch looks out of date.

timsifive, 2017-02-16:
The main branch is 0.13. That should be the one that shows up in github by default.

closed 2017-02-16 by mwachs5