riscv/riscv-debug-spec #26: "Debug mode" should be changed to "Halt mode".

poweihuang17, 2017-02-22:
When I reviewed the reset in 8.3, the term Debug Mode showed up:  
"If the halt signal is asserted when a core comes out of reset, the core must enter Debug Mode before executing any instructions "
However, the "debug mode" should be "halt mode".
So, every reference to debug mode should be changed to halt mode.
Moreover, the privileged spec use the term "debug mode" so we will need to sync with privileged spec for the term.

timsifive, 2017-02-22:
I think I fixed the debug spec. I'd like to hold off on changing the privileged spec until we're happy with the state of the debug spec.

poweihuang17, 2017-02-22:
yes, I feel the same for the privileged part. Thanks for your fixing.

mwachs5, 2017-02-22:
We could also avoid changing the priv. spec by saying something like "Halt
Mode is synonymous with Debug Mode described in Priv Spec"

timsifive, 2017-02-22:
On Wed, Feb 22, 2017 at 7:49 AM, Megan Wachs <notifications@github.com>
wrote:

> We could also avoid changing the priv. spec by saying something like "Halt
> Mode is synonymous with Debug Mode described in Priv Spec"
>

That seems unnecessarily confusing, especially since the Debug Mode
description in the priv spec was created by reading an older version of the
debug spec.

mwachs5, 2017-03-02:
@aswaterman FYI we are talking about removing the concept of "Debug Mode" entirely from this spec, it's just called "Halt Mode".

aswaterman, 2017-03-02:
The RISC-V privileged architecture refers to privilege modes by a single letter, and there's already an H-mode, for hypervisor.  So this name collision is not ideal.

Specifically, in the SiFive implementation, the additional privilege mode used by the debugger should continue to be named D-mode.  Or at least something that doesn't begin with U, S, H, M, or V.

wallento, 2017-03-08:
I am wondering: Why exactly did we name it "halt mode" and not "debug mode"?

poweihuang17, 2017-03-08:
I don't know, so I suggest we could change it from now.

mwachs5, 2017-04-05:
Thanks this has been changed.

closed 2017-04-05 by mwachs5