riscv/riscv-debug-spec #36: Two typo or problems of description

poweihuang17, 2017-03-02:
1. Table of page 40 is out of boundary and it's probably a latex problem.
2. In page 29 for serial port, the spec said "Overflow state is sticky, and can be reset by writing 0 to this bit.". What does it mean by sticky?

mwachs5, 2017-03-02:
ssue 1 is the same as #3 , right?

Sticky means that it can *only* be reset by writing 0 to this bit. I think we should change all these "write 0 to clear" to be "write 1 to clear" as noted in #33. But at the same time we can clarify the meaning of "sticky".

poweihuang17, 2017-03-06:
Yes, thanks for your reply.

poweihuang17, 2017-03-06:
I agree with "write 1 to clear", because "write 0 to clear" is too awkward.

mwachs5, 2017-03-06:
38cb5a0b  should clarify this. Let me know if the rewording makes sense.

closed 2017-03-10 by mwachs5