riscv/riscv-debug-spec #40: Define numbers for "nonstandard" versions

mwachs5, 2017-03-02:
From the mailing list (@asb):

Currently, the concept of the 'version' of debug support is accessible
via the version field in dmcontrol, the version field in dtmcontrol,
and xdebugver in in dcsr.

I propose that in all of those cases, we reserve and specify a version
number to indicate "non-standard debug is present". If nothing else,
this allows debug software to give a more useful error message.

mwachs5, 2017-03-22:
@asb did you have a more concrete proposal for this? Is this something done in the ISA which we can repeat the pattern?

timsifive, 2017-07-26:
I can think of several variations of "nonstandard":
1. There is some extra functionality added. In that case it makes sense to report the version that you are compatible with, and have some other mechanism to discover/access the extra functionality.
2. Your debug implementation is so different that it doesn't make sense to use the spec. In that case, why do you even have those version numbers in that place?
3. You're using parts of the spec (eg. the JTAG transport to access a Debug Module) but not others (eg. the JTAG transport). In that case you might need a special version in the Debug Module indicating that even though the transport is compatible, the rest of the debug system is not.

Not super important, but if we want people to be able to pick and choose different layers in our debug solution, nice to have. Easiest way is to just spec it and be done. It's cheap.

closed 2017-08-02 by timsifive