riscv/riscv-debug-spec #43: Clarify MISA during Debug Mode

mwachs5, 2017-03-08:
Should Hart use MISA during debug mode? Or should it ignore it and always just use the "most capable" version of itself. The latter means debugger would always get a consistent view of the hart's capabilities and not deal with MISA changing.

I think it's useful for MISA to at least report its current contents accurately, otherwise debugging is hard, so "save and restore"  in HW is a less appealing option. 

The other option is that Debuggers can write MISA to enable all capabilities whenever hart is halted, or whenever something that it expected to work fails, etc etc. 

Note that OpenOCD already does something similar for reading/writing the FPRs. In order to read them when hart is halted, it has to enable the bit in MSTATUS (or something) that enables the FPU. When the hart is resumed, it restores MSTATUS to its value.

So this is really a HW/SW tradeoff question. Because there are lots of simlar places where this is an issue (see above), I think it makes sense to keep it in the SW, but I could be convinced otherwise.

mwachs5, 2017-03-10:
I have updated the "Enumeration" document with the intention that MISA should act as if the machine was in Machine mode, there is no special debug interpretation of MISA.

closed 2017-03-10 by mwachs5

asb, 2017-03-11:
Excellent, that makes things easier.