riscv/riscv-debug-spec #53: File `jtag_registers.tex.inc' not found.

kito-cheng, 2017-04-05:
Hash: https://github.com/riscv/riscv-debug-spec/commit/9f3ec7ed08e05eabfd9ab83f3da1c017be11e346

```
...
(./hwbp_registers.tex.inc) (./core_registers.tex.inc)

! LaTeX Error: File `jtag_registers.tex.inc' not found.

Type X to quit or <RETURN> to proceed,
or enter new name. (Default extension: tex.inc)
```

sorear, 2017-04-05:
I hit this yesterday.  What happened is that I was missing pygments, so one of the Makefile rules failed, but it failed in a way that left behind some files, so the rule didn't run properly.  Do you get the same error on a clean build?

mwachs5, 2017-04-05:
I was wondering the same thing... the necessary file is there.

closed 2017-04-17 by mwachs5