riscv/riscv-debug-spec #55: Get rid of Resumereq race condition

mwachs5, 2017-04-06:
Consider the case where a debugger would like to single step. The spec implies that they should:

1) Set 'step' bit in DCSR
2) Resume the hart
3) Wait for the hart to be halted again
4) Check DCSR to see if it was because of single-step

The problem with this is that between steps 2 and 3, the debugger should really wait for the hart to have definitely resumed before checking DCSR. It may have just not resumed yet.  We really should have a bit which the SW writes and the HW clears upon resume. This could be resumereq, but really it should be another pair of status bits:

`resumedany`
`resumedall`

Is it sufficient to just have these bits cleared by the write to `resumereq`?

mwachs5, 2017-04-06:
I've added the description of these bits to DMSTATUS register.

palmer-dabbelt, 2017-04-06:
I think these should be called "resumeack", not "resumed", and that the semantics should be that the bits are cleared when "resumereq" is asserted and then set when at least one instruction has committed (on any or all harts).

mwachs5, 2017-04-06:
Fine with the naming... But one non-debug-mode instruction may never commit even if you do resume. You may have had a breakpoint on the first instruction. So I think it should be that basically it's OK to try to/ check for halt again now. If, on the same cycle, you read that resumeack = 1 and halted = 1, the hart has halted again.

palmer-dabbelt, 2017-04-06:
OK, that works -- I was just worried the language in there was too mushy.

mwachs5, 2017-04-06:
lmk what you think if  082c499b can be clarified

palmer-dabbelt, 2017-04-10:
Looks good.  I'm going to close the issue, as I think this is fixed.

closed 2017-04-10 by palmer-dabbelt