riscv/riscv-debug-spec #57: Quick Access -- is it useful

mwachs5, 2017-04-17:
I think as it is the Quick Access Abstract Command is not useful. There is a race condition that is not handled -- what if a hart halts due to a breakpoint? Do you really intend it to resume? Is the Quick Access command then an error?

mwachs5, 2017-04-26:
The concept of Quick Access only makes sense with a programmatic way of indicating whether you want to resume or not at the end of the program buffer. Right now there is no way to do that, debugger can only request resume by setting 'resumeReq' bit, or issuing Quick Access command which *always* resumes.

The problem is there may be some condition you want to branch on once halted (specifically DCSR.cause). We could list out the actual conditions and abstract them away with specific Debug Module bits on how that would work, or give some programmatic way to the program buffer to indicate that it wants to resume (or not) at the end of execution. In previous versions of the spec, this was accomplished with a combination of (a) The  DCSR.halt bit--- as long as this bit was set, the hart would remain in debug mode (conversely, if it wasn't set, the hart would just resume after executing program buffer) and (b) A set location for the 'resume' instructions in the debug ROM.

I think we don't really need the latter. Just having a bit with a defined meaning in the DCSR would be sufficient. Program buffer code could use it accordingly.

Then I would suggest getting rid of Quick Access abstract command, and adding a "execreq" bit to DMCONTROL.

Debugger would set 'haltreq' and 'execreq' at the same time. This is like a short cut to writing COMMAND register once hart is halted. Debugger causes hart to halt and immediately execute the program buffer. Program Buffer code is responsible for the TBD how to do "resume when done".

I'll keep thinking about this.

mwachs5, 2017-06-05:
Maybe this is not an issue, because if you resume off a breakpoint you would just break again. Need to verify that this is true.

mwachs5, 2017-07-04:
Still thinking about this. Resuming off a breakpoint you would NOT necessarily break again. The breakpoint may have been a data-match trigger and the read data may not be the same if read again (for example). It's actually unclear what would happen in that case in terms of resuming on an instruction with a read-data match. The processor would have to assume you want to re-execute the instruction and the data match may no longer be true.

timsifive, 2017-07-05:
That's a good point. More generally, it "just works" only if the system isn't modified while the quick access is being performed. Another hart could also remove the swbp that you just hit, for instance. These feel like pretty unlikely corner cases, but we should spec something robust that can deal with them.

mwachs5, 2017-07-05:
The spec currently says that "it is an error if the hart is already halted". Perhaps this covers this case and just needs clarification? 

If the hart is halted for some other reason than the debugger asking it to halt, then it's an error and shouldn't do the quick access.

timsifive, 2017-07-05:
That sounds like a good way to handle the issue. Quick Accesses aren't permitted while the hart is halted, so if they're attempted and lose the race, the command fails. It would be good to have a mechanism to explicitly say that the failure is because the hart is halted, otherwise there's still a race: try quick-access that fails, then the hart halts, then the debugger discovers the hart is halted and concludes the quick access failed because the hart is currently halted.

mwachs5, 2017-07-05:
Presumably busy/cmderr would specify that the quick access was still going / there was an error, respectively. Neither of these would be set in the case where the hart does the command then halts.

timsifive, 2017-07-06:
I would expect busy to be clear, since the command is no longer executing, but I'd change cmderr as such:
4 (running): The abstract command couldn't execute because the hart was running when it was attempted.
5 (halted): The abstract command couldn't execute because the hart was halted when it was attempted.

mwachs5, 2017-07-07:
A problem with the proposal that "if DCSR.cause != haltreq" is that the haltreq takes priority in DCSR. So even if breakpoint was hit on same cycle as haltreq, the breakpoint could still be skipped in this scheme.

timsifive, 2017-07-08:
I'm confused. Does this relate to Quick Access? AFAICT Quick Access is a separate abstract command, and does not involve haltreq. (At least if it does it's an implementation detail.)

mwachs5, 2017-07-08:
I am referring to my earlier comment, "If the hart is halted for some other reason than the debugger asking it to halt, then it's an error and shouldn't do the quick access." I was assuming "do quick access" included the debugger requesting the hart to "enter debug mode" even though you're right that the current spec says that DCSR.cause means specifically 'haltreq was set' not "the debugger asked the hart to enter debug mode." What would you expect DCSR.cause to be for quick access? I would have expected it to be "halt was requested by the debugger".

timsifive, 2017-07-09:
Ah! I understand now.
I agree that DCSR.cause should be "halt was requested by the debugger" while the hart is halted for the Quick Access.

In the case you're talking about, a Quick Access is attempted at the exact same time that a breakpoint instruction is executed. To make this simple, the core should serialize these debug requests. That seems reasonable since we're not going to enter Debug Mode partway through a multi-cycle divide either. Which one goes first doesn't seem important. Either the breakpoint instruction causes the halt and the Quick Access fails because the hart is halted, or the Quick Access causes the halt and the breakpoint is hit when the Quick Access has completed.

(In this case the priorities in DCSR.cause aren't important because there can only be one reason for the halt. Whereas it is possible for halt to happen on a single instruction because of single step, and it's a software breakpoint, and a hardware trigger fires.)

mwachs5, 2017-07-09:
Yes, that makes sense.

mwachs5, 2017-07-09:
Ok to summarize then I think these are the minor things that need clarification in the spec:

* DCSR.cause of 'haltreq was set' should be reworded as "debugger requested hart to enter debug mode"
* Clarify the quick access error case of  "it is an error if the hart is already halted" of quick access to "if the hart enters debug mode for a reason other than the debugger requesting it to do so". 
* Clarify how busy/cmderr are updated for quick access. We currently have on errror code of "halt/resume". Why do you think this is not sufficient?

I also still don't understand the Quick Access error description about "it is an error if the hart already resumed". How is the hart going to resume during the quick access? We have specified no way for a hart to resume itself.

mwachs5, 2017-07-11:
I think the clarifications make this implementable now.

closed 2017-07-11 by mwachs5