riscv/riscv-debug-spec #60: Resume "should" use resumereq and halt "should" use haltreq

poweihuang17, 2017-05-01:
In my opinion, the description for haltreq and resumereq is not clear enough now.
They mention that haltreq could halt the core when it's set to 1, but what happen when it is set to 0? 
Unsetting haltreq actually won't make the core resume,  and we need the resumereq to resume. However, this fact is not written on the spec, and we probably need to clarify that.
How does everyone think?

mwachs5, 2017-05-01:
If it's not clear, we need to clarify it!

Right now the description is here: https://github.com/riscv/riscv-debug-spec/blob/0.13/debug_module.tex#L101

and here:
https://github.com/riscv/riscv-debug-spec/blob/0.13/xml/dm_registers.xml#L89

The intent is that once a hart is halted, it doesn't matter what you write to haltreq (1 or 0). But, if the hart runs again (because you did a resume, or reset) then what is contained in haltreq matters, so you should write it to 0 before doing resumereq (it does say that is undefined), or 1 or 0 during reset depending on what you want to happen when hart comes out of reset.

poweihuang17, 2017-05-02:
Thanks for your quick reply. I only looked at the table before. After I see the description in 3.4, it's quite clear to me now. However,

mwachs5, 2017-05-05:
Yes, I think the statement that "resumereq has no effect on hart which is running" is probably the wrong part. I think it should say that debuggers should not do this, because its effect on harts which are running or will halt in the future is undefined. What do you think?

Also, there was another question on the mailing list about this section. I realized the part that is in a comment might be read as referring to what debugger should do on both halt and resume, when it really only means what it should do on resume:
https://github.com/riscv/riscv-debug-spec/blob/0.13/debug_module.tex#L121. We should either clarify or remove this comment.

mwachs5, 2017-05-06:
I have attempted to make this more clear in the latest version of the spec.

closed 2017-05-06 by mwachs5