riscv/riscv-debug-spec #67: sw_registers.tex is missing

palmer-dabbelt, 2017-05-10:
````
$ make -j1
make: *** No rule to make target 'sw_registers.tex', needed by 'riscv-debug-spec.pdf'.  Stop.
````

It looks like this is from 0512f5d9e9f80b392b91ca32c17f493544820302.

donbmouse, 2017-05-10:
Duplicate of #61.

closed 2017-05-10 by palmer-dabbelt

mwachs5, 2017-05-11:
yeah sorry can't get onto github right now. Can you just 'touch' the file
and add it for now?

palmer-dabbelt, 2017-05-11:
I tried already, it blows up.  Where is your workdir?

mwachs5, 2017-05-11:
Ok sorry it's on my laptop. Can you just do a "revert" of the change that
added that file through github? I realize I forgot to actually delete the
old code anyway so no information is lost.

mwachs5, 2017-05-11:
Wait never mind, I'll just make a PR that actually fixes the issue.