riscv/riscv-debug-spec #70: Support more than 1024 harts

mwachs5, 2017-05-15:
Feedback from the workshop

timsifive, 2017-07-26:
This might actually be easy:
If a system has more than 1024 harts, the Debug Module must support hawindowsel and hawindow. hawindowsel will be as wide as is required to be able to access every hart.

mwachs5, 2017-07-26:
So 32*32 is the actual limit then?

mwachs5, 2017-07-26:
oh wait no its 2^32 * 32. That seems... OK.

sorear, 2017-07-27:
Possibly relevant: the current [1024-way read mux for halt flags](https://github.com/freechipsproject/rocket-chip/blob/9804bdc34e81ae6e1674a7c0102480f167763443/src/main/scala/devices/debug/Debug.scala#L888) was responsible for [most](https://gist.github.com/sorear/eca052c170c7b754b9f0f06af968388e) of the non-TLMonitor rocket-chip firrtl output when I last did a build a couple months ago.

timsifive, 2017-08-02:
Continuing this thought, do we need hartsel at all?
When there is just one hart, hawindowsel would be tied to 0, and hawindow tied to 1.
On systems with 32 harts or less hawindowsel would still be tied to 0. hawindow would use its full 32 bits. That's a bit more expensive than the 5 bits otherwise required by hartsel, but if you have that many harts I think you can spare those bits.

The only issue is whether it should be possible to set multiple bits at once or not. hawindow could be RAWL (did I get that right?) to deal with that.

mwachs5, 2017-08-02:
it is nice to be able to set halt/resume on a particular hart without ambiguity/race.

(that being said, it causes implementation pain to allow hartsel being in the same register as haltreq)

timsifive, 2017-08-02:
> it is nice to be able to set halt/resume on a particular hart without ambiguity/race.

I agree, but I'm not sure what exactly this comment applies to. If the debugger wants to do that, it should only set a single bit in hawindow.

mwachs5, 2017-08-02:
right, but that takes two seperate writes

timsifive, 2017-08-03:
That's true. Is that the only benefit of the hartsel method? That you can change harts and request halt/resume all in one write? That doesn't really seem worth having a separate hart selection mechanism for.