riscv/riscv-debug-spec #98: Describe our probing sequence

palmer-dabbelt, 2017-06-13:
The trickiest part of our OpenOCD port is figuring out how to probe the system from nothing -- this is what enforces many of the constraints on what implementations we support.  We should write this down in the manual somewhere once we've got it all correct.  Essentially it

 * Does "csrw dscratch", "auipc", "sw", "csrr dscratch" to figure out where the progbuf is, and if the progbuf is writable.  We don't support non-writable progbufs, but we could easily.
 * [not implemented yet] Does "addw x0, x0, x0" to see if 64-bit is supported.
 * Does another AUIPC sequence, but with a "sd" instead of "sw".  This way we can tell where the full address of the progbuf is.

At this point we don't check for FP or which CSRs are valid, so we need some mechanism for that as well.

mwachs5, 2017-06-13:
Wouldn't that first bullet have the same problem that dscratch isn't restored if it fails?

palmer-dabbelt, 2017-06-13:
The fix would be to just look at progbuf size and make sure we're always writing to the data words.  If neither are writable then you're right it would corrupt the register, but we wouldn't actually be able to debug anything.

IIRC there's a way to ask if the data words are memory mapped?  We could check that and say "unsupported", at least it won't corrupt the target.  The other option would be to install a trap handler that restores dscratch, but then we'd need a second scratch register and a handful of extra instructions.

mwachs5, 2017-06-13:
Yeah there is hartinfo reg to tell you they are memory mapped.

timsifive, 2017-09-13:
Per Nikhil, be sure to address:
> Sec 3.5, intro section
> "Table 3.2: Use of Data Registers"
>
> It would be good to discuss how the debugger discovers XLEN, so that
> it can know how to interpret data0, data1, etc.
>
> The debugger could read CSR 'misa', in which the upper two bits
> (misa.MXL field) encodes XLEN.  But there is still a problem.  When
> XLEN=32, the upper two bits of misa will be in data0[31:30}, whereas
> when XLEN=64, the upper two bits of misa will be in data1[63:62].
>
> Also, when XLEN=64, misa[31:30] is within a "WIRI" field, so an
> implementation could legally return a 2'b01 in data0 [31:30], giving
> the debugger a mistaken impression that XLEN=32.
>
> How is this resolved?

timsifive, 2019-11-25:
Closing due to lack of interest.

closed 2019-11-25 by timsifive