riscv/riscv-debug-spec #99: Does COMMAND register need to be fully R/W

mwachs5, 2017-06-27:
Right now we don't specify usage for most of the bits in COMMAND. Must the register hold all bits, even those that we currently have no definition for? I think that in keeping with most of the RISC-V spec,  some bits may be WARL. I think COMMAND.type is the only field which is truly R/W.

timsifive, 2017-08-02:
Does the command register need to be readable at all? It might be nice for debugging the debug logic, but I don't think an external debugger cares at all. There already is a separate mechanism for indicating that a requested command was not supported.

timsifive, 2018-01-09:
Right now, the cmtype and control fields (which between them make up the entire command register) are listed as write-only. Is anything more needed?

mwachs5, 2018-01-09:
I think that's fine

closed 2018-01-09 by timsifive