riscv/riscv-debug-spec #103: Sync with user level spec about ebreak

poweihuang17, 2017-07-01:
Hi,
I'm summarizing the functionality of ebreak, and I found some points that may need clarification.
Firstly, Ebreak actually has 2 usage in debug spec, but the user level spec only cover one .
Functionality of ebreak in debug spec:
1. Enter into debug mode when ebreak is executed. (ebreakm,ebreaku,ebreakh)
2. Mark as the end of program in program buffer.
However, in user spec, only the second one is covered.
"The EBREAK instruction is used by debuggers to cause control to be transferred back to a debugging environment"
So, the user spec needs change.
---------
Besides, why do we need to have a new instruction ebreak? (Hope it's not a repitition)
For the first part, why don't we just use ecall and assign a specific parameter for debug exception?
For instance, the x86 use "int 3" for debug exception, so there is no need for a new instruction for transferring  into debug mode.
For the second part, ebreak serves like a EOF, but we have a "progsize" parameter in debug  spec. So, why don't we just add "NOP" at the end of program and end the program buffer when it counts from 0 to progsize?
----------
The purpose of these question is to answer what does ebreak do, and why do we design a new instruction to achieve these functionality.  Thanks for the help.

mwachs5, 2017-07-03:
The user spec does not need to describe what happens when not in user mode. The secondary usage of ebreak only defines what ebreak does in debug mode. From the user mode perspective, ebreak does just cause an ebreak exception.

mwachs5, 2017-07-03:
"Why do we need to have ebreak vs ecall?" -->

These are two exception types which are individually controllable and
delegable by both debugger and machine mode. ecall is generally a call into
a more privileged mode, while ebreak is generally a call into the debugger.
One can be delegated without the other, and debugger can take over ebreak
handling without affecting ecall handling.

For the second question of "why not just end the program buffer when it
counts from 0 to progsize" -- that prevents you from implementing the
program buffer in some general purpose SRAM, unless some words of that SRAM
were hard-baked to be "jump back to start".

Megan

mwachs5, 2017-07-10:
@aswaterman Can you confirm that it is OK to have this secondary meaning of ebreak in the debug spec without updating User spec to include it

aswaterman, 2017-07-11:
I concur that the user ISA spec should not change.  The alternate behavior of EBREAK in debug mode is not visible to user mode, and so it should not be described in the user ISA.

poweihuang17, 2017-07-11:
Thanks for the reply and sorry for the late reply from myself. I agree with both of you. But I have a question about this sentence: "The EBREAK instruction is used by debuggers to cause control to be transferred back to a debugging environment".
Does it equals to "Enter into debug mode when ebreak is executed. "  ?
What does "debugging environment" refer to?

mwachs5, 2017-07-11:
Ebreak only enters debug mode if there is an external debugger present. Really what it does is cause a breakpoint exception. DCSR (writable only by Debugger) controls what happens on a breakpoint exception.

 Usually, with no debugger attached, debug environments running in m,s,or u code could replace instructions in user code with ebreak, as a software breakpoint. M mode can transfer control to software debugger, or  delegate breakpoint to to S or U mode like other exceptions. The code doesn't HAVE to be a debugging environment, but that's what it's there for.

aswaterman, 2017-07-11:
Yeah, it is deliberately vague in the user ISA spec because the meaning of
"debugging environment" depends on the privileged architecture, so isn't
part of the user ISA.

poweihuang17, 2017-07-12:
ok. I'll close this issue. But sometimes I couldn't easily know what those phrase like "debugging environment" refer to. I will give a pull request if I find better way to rephrase. Thanks.

closed 2017-07-12 by poweihuang17