riscv/riscv-debug-spec #120: Should we declare that ndmreset assertion time must be synchronized

mwachs5, 2017-07-26:
Clarify that that implementer needs to make sure that this reset is synchronized to their system, and asserted as long as possible.

Or should it be a 

Or is there a reset handshake -- we poll to see if it is out of reset.

timsifive, 2017-07-27:
Brainstormy thoughts:
1. We could add a register that indicates in real time what the ndmreset/reset pulse widths should be.
2. If there is some kind of system ready signal internal to a system, it would be cool to expose that in dmstatus.

timsifive, 2017-09-13:
Nikhil's feedback raises this issue as well:
>  - Resets have no handshake: the debugger asserts and then deasserts
>       reset after some fixed number of cycles (even though different
>       implementations may take varying number of cycles to perform
>       resets).

timsifive, 2017-09-20:
From Nikhil:
> Section 3.10, Section on dmcontrol: hartreset, ndmreset and dmactive
> "the debugger writes 1, and then writes 0 to deassert the reset"
>
> Why? The debugger does not know how long a reset takes, so it's dicey
> to count time or cycles.  Why not make these R/W1 fields, instead of
> R/W, and let the debugger poll for hartresetack, ndmresetack and
> dmactiveack, respectively, either at these same bit positions, or in
> separate bits in the dmstatus register?

mwachs5, 2017-10-16:
To answer the question of "Why do we have to write 0 to deassert the reset" --- asserting reset may kill all the clocks in the system, except the debug/test interface clock. It's not obvious what logic would deassert that reset. We could say "after 1 debug module interface clock tick it is automatically reset" but that introduces the idea of a Debug module interface clock, which is pretty low-level implementation detail. Saying "you have to write 0 to deassert reset" is clear and straightforward to implement.

But we should still clarify that the hart(s) may not ACTUALLY come out of reset for some time later, again this is also described in #165

mwachs5, 2017-11-17:
Closed by #172

closed 2017-11-17 by mwachs5