riscv/riscv-debug-spec #121: add Single Step section that details some corner cases

timsifive, 2017-07-26:
> For example, let's say our $mtvec is set to 0xDEAD_BEEF (some bad non-executable address).
> 
> In debug mode, we set $dpc to 0x0 (some non-executable address).
> Then we set $dcsr.step and resume.
> 
> When we exit debug mode, we will immediately get an exception and immediately enter debug mode, because you're supposed to do that when you get an exception in single step. $mepc should be updated with 0x0. Presumably $dpc holds 0xDEAD_BEEF?
> 
> I feel like this needs to be clear since "after the one that was stepped over" doesn't really apply if you never even completed the instruction you stepped over.
> 

--Megan

mwachs5, 2017-07-26:
I think we did have some discussion on this already, need to revisit the text to see if this is clear

timsifive, 2017-07-26:
I can't find a Single Step section in the current version of the document (except in the implementation section). I feel it should be added to the RISC-V Debug section.

mwachs5, 2017-07-26:
I think this is the bit I was talking about:
https://github.com/riscv/riscv-debug-spec/blob/c1e61b0530085a6ed19b738de1585621decd26d1/xml/core_registers.xml#L73

mwachs5, 2017-07-26:
I would say that that linked text doesn't make it clear what $dpc would be set to in that case (or in all cases)

timsifive, 2017-07-27:
I agree, but I don't think that is a good spot to delve into the corner cases of single step, just because it's in a table and formatting wise it will look wrong.

closed 2017-12-13 by mwachs5