riscv/riscv-debug-spec #137: Is 4 bits enough for the version number in DMSTATUS?

poweihuang17, 2017-08-23:


timsifive, 2017-08-24:
The usual solution is that when we get to version 15, we add bits somewhere else and then say that if this field is 15 then to look there. My hope is that it will be quite some time before we get that far.

poweihuang17, 2017-08-25:
Yeah. It could be. It's an issue that Rishiyur pointed out in the meeting. He doesn't want the 4 bit to be a constraint in the future, and the register do have place to increase from 4 bits. I think your solution will be better, as it's a software approach.
Just add a feedback from what I was asked when I'm interviewed by MediaTek folks for a graduate level job. After I tell them about all the implementation, including Syntacore(0.90), SiFive(0.13), ???(0.11), Bluespec (Custom approach), they're wondering why there are so many versions given that we have a standard. I replied it by a compatibility of gdb, but again, if we're satiesfied with gdb level compatibility, the meaning of the standard would be ambiguous. I guess in the end it will be better for all 3rd party companies and IP companies to resolve to less than 4 versions.

timsifive, 2017-09-30:
Rishiyur seems satisfied with this explanation as well. E-mail from:"Rishiyur Nikhil" subject:"[Debug] Detailed comments on latest draft of debug spec":
> ...
> >   > Section 3.10, Section on dmstatus
> >   >
> >   > Are 4 bits enough for future evolution of the 'version' field? (dmstatus has many unused bits that could be reserved now for a wider 'version' field)
>
> >   When we get to version 15, we'll say that 15 means to look somewhere else for the true version. I don't foresee us getting there any time soon.
> 
> Ok.
> ...

Can we close this issue?

poweihuang17, 2017-10-01:
Yes. I think we could.

poweihuang17, 2017-10-01:
I'll close that now.

poweihuang17, 2017-12-11:
Subversion within 0.13 cause this issue to be reopened...
See Markus's email.

mwachs5, 2017-12-13:
related to #188

timsifive, 2018-10-26:
Closing due to lack of interest/activity.

closed 2018-10-26 by timsifive