riscv/riscv-debug-spec #139: inconsistent error protocol between abstractcs and sbcs

timsifive, 2017-09-13:
From Nikhil:
>  - Some handshakes involve commands in one register with error status
>       in another register (e.g., 'control' and 'abstractcs'). After an
>       error, the error has to be cleared in the latter register before
>       the next comand on the former register, requiring an extra
>       transaction from the debugger.
>
>  - Some handshakes involve commands and error status in the same
>       register (e.g., 'sbcs'). After an error, the error can be
>       cleared simultaneously with the next command.

Note that control's errors are completely different from both of those, and (I think) with good reason.

timsifive, 2017-11-22:
If we're making backwards incompatible changes to System Bus Access to address #150 then I'm fine also breaking up sbcs into sbcontrol and sbstatus. (I believe we have register space for it.) Otherwise I'm inclined to leave things as is.

timsifive, 2018-01-11:
I'm changing my mind on this.
abstractcs puts everything in one register. We also have dmcontrol and dmstatus. The reason the latter are split up is that all the bits don't fit in one register. I don't see a good reason to make a second register because in one case we were forced to.

mwachs5, 2018-01-11:
I agree, especially since we are already using up a lot of registers for SBaccess

closed 2018-01-12 by mwachs5