riscv/riscv-debug-spec #140: spec does not address cache coherency wrt system bus interface

timsifive, 2017-09-13:
From Nikhil:
> The spec may need some additional functionality to force coherence
> between the Debug Module's view of memory through a system bus master
> and the hart's view of memory through its caches.  This is analogous
> to the functionality of the FENCE and FENCE.I instructions.

I would prefer to not mandate any kind of caching behavior, but punt this problem to the debugger. It could use the program buffer for cache control instructions. Alternatively we could add abstract commands or system bus commands to perform cache operations as well.

timsifive, 2017-09-13:
Also from Nikhil:
> Is the spec missing some functionality to re-establish coherence?  For
> example:
>
> - When the debugger uses the system bus master to write to instruction
>     memory, and then resumes the hart, the hart must see the updated
>     instruction memory, i.e., it's caches must be 'coherent' with the
>     new contents of instruction memory.
>
> - When the hart halts, and the debugger then examines memory using the
>     system bus master, it should see contents 'coherent' with the
>     hart's caches.
>
> Should we provide debugger commands to force coherence?
>
> (This is probably not an issue with the Program Buffer method of
> accessing memory, since those accesses are done through the hart
> itself.)

mwachs5, 2017-11-16:
closed by #171

closed 2017-11-16 by mwachs5