riscv/riscv-debug-spec #142: Add page/section references to \R or \F macros

timsifive, 2017-09-13:
Currently \R macros provide hyperlinks to the register definition. Should they provide Section/page numbers also? Maybe just on the first mention? Same could be done for \F macros.

Nikhil:
> Sec 3.2, para 1
> "... while dmactive in dmcontrol is 0."
>
> Suggest rewording "... using the dmactive bit in the dmcontrol
> register (Section ...)." with a forward reference to the section where
> the dmcontrol register is described.

mwachs5, 2017-09-13:
that seems like it would be unwieldy in general... but perhaps we could generate another macro, like \REF, used when desired, like:

"See \Rdmcontrol (\REFdmcontrol) for more information".

what the \REF points to could be page, or Section, as set up by the configuration.

timsifive, 2017-09-20:
Also from Nikhil:
> Sec 3.5.1, "Access Register" section, bullet point 1
> "... if write is clear and transfer is set"
>
> Suggest: place a forward reference to 'dmcontrol's description in Sec
> 3.10, page 17, which is where 'write' and 'transfer' are described.

timsifive, 2017-09-27:
Also from Rishiyur Nikhil:
> Section 4.4, on dpc: Table 4.2
> "if timing is 0 ..."
>
> Insert forward ref to where 'timing' is defined, in Sec 5.1,
> 'mcontrol' register.

closed 2017-11-03 by timsifive