riscv/riscv-debug-spec #143: Can resumereq be W1?

timsifive, 2017-09-13:
Nikhil:
> Sec 3.4, para 2
> "... before clearing resumereq to 0"
>
> Why not make resumereq a W1 field instead of a W/R field, so that you
> don't have to clear it to 0?

I think the answer is that it makes sense for haltreq to be level sensitive, and to be consistent then we also have resumereq be level sensitive.

timsifive, 2017-09-13:
Also:
> Sec 3.4, para 3
> "Setting resumereq on a hart which is running ..."
>
> See previous comment (Sec 3.4 para 2) on suggestion to make resumereq
> a W1 field.  In which case, the DM can simply ignore this if the hart
> is already running.

mwachs5, 2017-09-13:
To elaborate, haltreq should be level sensitive so that you can request harts to halt across reset... you set all their halt bits (by writing haltreq =1 for each hartsel value), then asserting reset. If it was W1, it would be ambiguous whether you want the haltreq to persist across reset.

mwachs5, 2017-11-08:
I believe this was clarified by #158

closed 2017-11-08 by mwachs5