riscv/riscv-debug-spec #144: move Abstract Command Listing next to \Rcommand

timsifive, 2017-09-13:
Nikhil:
> Section 3.5.1, General
> "Abstract Command Listing"
>
> This section may be better placed closer (e.g., as a subordinate
> section) to the 'Abstract Command' section in 3.10.  It's easier that
> way as a reference manual, avoiding page flipping back and forth.

That makes sense, but LaTeX (and our dropping of subsubsections) makes this tricky to get the details right.

mwachs5, 2017-09-13:
we can tweak the format if you want subsubsections, I have no problem with that. I just copied the other RISC-V manual format. 

BTW I think you can use [h] for "here" to ask LaTeX to put things "here".

timsifive, 2017-10-18:
I think this was done in 523c64af.