riscv/riscv-debug-spec #150: figure out sbsingleread, sbautoread

timsifive, 2017-09-20:
There are 4 use cases, and I don't think they're best served by what we have:
1. Read from consecutive addresses. We'd like to:
   1. Write address (triggering read)
   2. Read data (triggering read+postincrement), ...
2. Read from non-consecutive addresses.
   1. Write address (triggering read)
   2. Read data
   3. Repeat
3. Write to consecutive addresses.
   1. Write address
   2. Write data (triggering write+postincrement), ...
4. Write to non-consecutive addresses.
   1. Write address
   2. Write data (triggering write)
   3. Repeat

The bits we need to achieve that:
1. postincrement
2. Trigger read on address write (addresstrigger?)
3. Trigger read on data read and write on data write (datatrigger?)

Does that sound right?

mwachs5, 2017-09-26:
Those sound like reasonable use cases. How much overhead is introduced by the current solution?

timsifive, 2017-09-26:
In our current spec:

1. Read from consecutive addresses.
   1. Write address
   2. Set sbautoread+sbautincrement
   3. Read data N-1 times (triggering increment+read)
   4. Disable sbautoread
   5. Read data
2. Read from non-consecutive addresses.
   1. Write address
   2. Set sbsingleread
   3. Read data
   4. Repeat...
3. Write to consecutive addresses
   1. Set sbautoincrement
   2. Write address
   3. Write N words to data (triggering write+increment)
4. Write to non-consecutive addresses.
   1. Write address
   2. Write data (triggering write)
   3. Repeat

So the reading is a bit more clunky than necessary right now.

mwachs5, 2017-09-28:
In the proposed solution I think there is something missing, you still have to write some control bits which would say whether you want the auto read & increment behavior to happen, right?

I think you should perhaps keep the current bits autoread and autowrite names, and just slightly change their functionality -- for read when it is set, you want the read to happen on address write or data read. For autowrite, you just want the write to happen on data writes.

timsifive, 2017-09-29:
Yes. The "proposed solution" was really a wishlist, and doesn't include the setup. Each one of them (probably) will require a single setup write at the start.

There is no autowrite bit in the current spec. It's the standard behavior, and could be in the new version as well.

The current bits are: sbsingleread, sbautoincrement, sbautoread
Proposed bits are: sbautoincrement, sbreadonaddr, sbreadondata

sbautoread is the same as sbreadondata.

sbreadonaddr replaces sbsingleread. It's more parallel to sbreadondata so hopefully easier to understand. It also allows for more efficient reading at non-consecutive addresses, although that's a pretty rare use case.

mwachs5, 2017-12-05:
@albert-magyar, as someone who presented an implementation of the system bus interface at the last workshop, any thoughts?

mwachs5, 2018-01-09:
I think part of the problem is we're rubbing against the informal RISC-V rule that things shouldn't be ratified until implemented.  Neither rocket-chip, nor https://github.com/google/bottlerocket, nor riscv-isa-sim (Spike) have implemented the system bus access feature. I don't know of any open-source implementations which have which could serve as a reference/validation of either protocol.

Does this mean that we need to move all reference of System Bus access from the ratifiable portion? That would be a shame since it is a useful, oft-requested feature. Is it possible to implement this feature in Spike at least?

Implementing in rocket-chip will happen eventually, but I can't provide a short-term timeframe for that.

albert-magyar, 2018-01-09:
I do think `sbreadonaddr` is a more intuitive complement to `sbreadondata` than single/auto.

Unfortunately, my experience with System Bus Access is in using it:

1. In halt only (though this wasn't a functional requirement)
2. In the absence of caches or other bus masters

That made it reduce to a very simple FSM. It is not currently in the open-source version, since the bus master ports were changed in the open-sourcing process, but it would be very quick to add back. I do also need to write a standalone testbench for debug mode in the open-source repo, since it lacks an open-source TAP at the moment :)

timsifive, 2018-01-09:
@mwachs5 So whatever I implement in spike/OpenOCD goes? I do feel we've given people who have system bus implementations plenty of opportunity to comment here.

mwachs5, 2018-01-09:
Yes, I think so. If it's really a problem we can use the #188 and differentiate or as you said use the `sbcs` bits. But let's prioritize figuring out the right interface and then we can increment a version number if necessary. Perhaps the result of the experiment is that the old one is exactly what we wanted, but hard to say that conclusively.

timsifive, 2018-01-09:
We know that there are people who have implemented the existing bus master spec, so we will have to increment the version number.

poweihuang17, 2018-01-10:
@timsifive @mwachs5 Thanks for your notification. I've read through this issue and the pull request.
General observation made by me:
- [x]  From the example that Tim provided (section B), the quantity of operations are almost the same and hence the program size. 

 The pros:
- [x] The logic is clearer and parallel.

The cons:
- [x] Backward compatible change.
- [x] Address trigger consume more power consumption. Especially, when we want to read from the same address multiple times (memory mapped I/O), we will have to save the address(32 and more) so many time. But in original case, we only need to write 1bit (sbsingleread) many times. 

So, personally, I  don't favour to change.

poweihuang17, 2018-01-10:
@codelec Kritik, did you implement system bus access in librecore sodor? I used grep and found that there are some codes about sbsingleread in repo. Do you have anything to comment?

codelec, 2018-01-12:
I have implemented system bus access and I am using datatrigger i.e. on accessing sbdata register a memory access is performed.  I am not sure how addresstrigger will be helpful. For sodor's current use cases datatrigger is sufficient and better than addresstrigger

https://github.com/codelec/riscv-fesvr/blob/master/fesvr/dtm.cc
https://github.com/librecores/riscv-sodor/blob/arty-dev/src/common/debug.scala

timsifive, 2018-01-12:
@codelec Thanks for commenting.

sbreadonaddr is helpful when reading memory from non-consecutive addresses (which is pretty rare). This change is mostly about having a simple-to-understand interface.

mwachs5, 2018-01-12:
I haven't read through this updated proposal yet. Is it in "finished' state? Or still WIP?

timsifive, 2018-01-12:
@mwachs5 I thought it was finished, but Rishiyur has been finding some issues (that also existed in the old spec). It's fundamentally no different from what it's been for a few months.

mwachs5, 2018-01-12:
Thanks, I'll take a fresh look tonight then

mwachs5, 2018-01-17:
I removed the "backwards incompatible" label since the sbversion field in the proposal is covering that

timsifive, 2018-01-26:
Fixed by #203

closed 2018-01-26 by timsifive