riscv/riscv-debug-spec #151: is there one haltreq signal, or is there one bit per hart?

timsifive, 2017-09-20:
The spec is not entirely clear whether haltreq directly connects to only the currently selected harts, or if it modifies a bit which persists even when the currently selected harts changes. The former seems easier to implement in hardware. The latter would help resetting multicore systems that don't support hawindowsel.

Based on Nikhil's observation:
> Section 4.3, para 1
> "If the halt signal is asserted when a core comes out of reset, the
> core must enter Debug Mode before executing any instructions, ..."
>
> Clarify:
>
> "If 'dmcontrol.haltreq' is asserted when a core comes out of reset,
>  then all harts selected by dmcontrol.hasel/hartsel must enter Debug
>  Mode before executing any instruction ..."
>
> and
>
> "If 'dmcontrol.haltreq' is asserted when a hart comes out of reset,
>  then if it is selected by dmcontrol.hasel/hartsel, it must enter
>  Debug Mode before executing any instruction ..."

mwachs5, 2017-09-20:
The latter is *required* for resetting multicore systems that don't support hawindowsel. Let's clarify the intention in the text.

timsifive, 2017-10-05:
What happens on a 2-hart system, when you:
1. Select hart 0, and write haltreq
2. Select hart 1, and clear haltreq
3. Select both harts, and read haltreq

I'm tempted to make haltreq, resumereq and hartreset write-only.

mwachs5, 2017-10-05:
Yes, it was already sort of ambiguous whether they were readable.

closed 2017-11-02 by timsifive