riscv/riscv-debug-spec #153: Should we have a separate cause for halted-out-of-reset?

timsifive, 2017-09-27:
From Rishiyur Nikhil:

> Section 4.4, section on dcsr: cause field
>
> Would it be useful to distinguish the following with another code?
>     3 [existing]: haltreq was set while the hart was running
>     5 [proposed]: haltreq was set when the hart came out of reset

Advantage: Debugger could tell whether its attempt to halt out of reset was successful or whether the hart simply halted after being reset.

Disadvantage: More complex hardware implementation.

They both feel minor.

mwachs5, 2017-09-28:
I don't think it's super trivial to implement [5]. What does the debugger user gain from this information? It is *required* that the hart halt out of reset if haltreq is set during reset, so I suppose the only race is if the debugger connected to a system whose reset being asserted unpredictably & out of the debuggers control, then it's a nice-to-know that the hart happened to just come out of reset?

mwachs5, 2017-10-16:
I think this may be somewhat addressed by #165

closed 2017-11-01 by mwachs5