riscv/riscv-debug-spec #165: Request for "has been reset" indicator

mwachs5, 2017-10-16:
(From Yung-Ching Hsiao) The current debug spec is lacking the capability to report the occurrences of resets (for the selected hart). Examples where this might be useful: While the external debugger is waiting for some breakpoint to happen: 

* watchdog fires and triggers reset. 
* Accidental touch of the reset button (pilot errors). 

The external debugger software could deal with these situations more gracefully if such information is available to it. I wonder if something like dmstatus.ANYRESET makes sense to everybody, or could we allow dmstatus.ANYUNAVAIL to be a sticky bit (and require W0 to dmstatus to clear it?).

mwachs5, 2017-10-16:
I think this is a good feature request. I think we would want to distinguish between "is currently unavailable" and "has been unavailable since last cleared". I wonder if we want to overload "unavailable" or do we really mean "has been reset"?

timsifive, 2017-10-18:
I lean towards anyreset/allreset bits in dmstatus, a clearreset bit in dmcontrol to clear the state. Whenever a hart resets for any reason it sets it reset signal, and the DM notices this and sets a sticky bit. Later the debugger comes along and clears it by writing clearreset.

How does that sound?

mwachs5, 2017-10-18:
I think the general approach is good, but some different namings may be less confusing. I'm copying your text below with some different names:

I lean towards *anyhasreset/allhasreset* bits in dmstatus, a *hasresetack* bit in dmcontrol to clear the state. Whenever a hart resets for any reason it sets it reset signal, and the DM notices this and sets a sticky bit. Later the debugger comes along and clears *the reset ack for any selected harts* by writing *hasresetack*.

closed 2017-11-01 by mwachs5