riscv/riscv-debug-spec #169: Are "conceptual" bits cleared on dmactive = 0

mwachs5, 2017-10-30:
we clarified the spec to explain that there are hart request, resume request, etc bits.  Also there are hasreset bits.

Do we need to clarify the impact of dmactive on these bits?

It seems like hart request, resume request, would be cleared to 0 when dmactive is 0, but that wouldn't really change the fact of whether the hart is halted or running or not. So maybe these are less confusing.

For hasreset, should those bits persist regardless of dmactive? This would also clarify that the first time a debugger hooks up, and sets dmactive to 1 for the first time, hasreset could be 0 even though the hardware was reset.

timsifive, 2017-10-30:
It doesn't matter to me what happens to those bits when dmactive goes low, so I'm inclined to spec whatever is simplest to implement. If a debugger cares about these bits, it should not let dmactive go low.

mwachs5, 2017-11-16:
Also the spec already says that the other bits aren't really defined when dmactive is 0. I'm fine with that.

mwachs5, 2017-11-17:
Closed by #172

closed 2017-11-17 by mwachs5