riscv/riscv-debug-spec #180: Access to unimplemented Debug Module Registers

zhonghochen, 2017-11-23:
https://github.com/riscv/riscv-debug-spec/blob/ce1b359329ba87c3302df496885ff3781cb4c053/debug_module.tex#L363-L364

Since dmerr is added, would it be better to update the description? 
Writing unimplemented Debug Module DMI Registers updates dmerr to 1.

mwachs5, 2017-11-27:
I agree that these two sections are somewhat inconsistent. But I don't see a good use of dmerr so far, and one way of "implementing" debug module registers is to return 0 and ignore writes if the functionality doesn't exist. This is pretty clearly spelled out for every register that is optional. So I am not sure what functionality dmerr is adding other than reserving space for future TBD  uses.

zhonghochen, 2017-11-27:
Could dmerr be optional? Is there any usage for justifying dmerr is worth to be implemented? 
Debugger requires an extra step for clearing dmerr after confirming a feature is not implemented.

mwachs5, 2017-11-27:
I agree with the above... requiring the debugger to both attempt to read the register, checking dmerr, and clearing dmerr, seems like much more work for both SW HW implementers than just reading the register and seeing if it has it's "unimplemented" value as defined in the spec.

mwachs5, 2017-12-18:
What if we change the text from:

```

timsifive, 2018-01-11:
Per Issue #190, we're about to remove dmerr.

closed 2018-01-11 by timsifive