riscv/riscv-debug-spec #186: Add Canonical sequences to appendix

mwachs5, 2017-12-13:
Currently we have some examples of sequences that will allow various types of access using both the Abstract Command, System Bus access, and Program Buffer. Can we flesh these out fully to include all the access types? And perhaps the necessary polling for "busy"?

timsifive, 2017-12-13:
It would be really nice if we can put these in OpenOCD, under a compile flag. That code would differ from the current code by having as its goal to be as readable as possible without regard to speed. By having it in OpenOCD we could verify that it's correct by running the existing debug test suite.

mwachs5, 2017-12-13:
So you mean that the appendix could be converted directly into C code sequences (similar to the header file)? Or just include C code snippets directly?

timsifive, 2017-12-13:
I'm not exactly sure. I worry that if it's just a human-written appendix then it becomes hard to test.
Another option might to use OpenOCD debug output and then annotate that.

mwachs5, 2017-12-13:
I think it's a good point. We would also need something to test against in that case... for example we would need a model to handle system bus access and abstract command GPR access (because I think the request was for something more than just Program Buffer examples)

timsifive, 2017-12-13:
Is this a requirement for ratification, or just a nice-to-have? It feels like a lot of work.

mwachs5, 2017-12-13:
I think it is a nice-to-have. I'm not even sure it is necessarily part of the spec. It's basically a reference that could go anywhere.

timsifive, 2019-11-25:
Closing due to lack of interest.

closed 2019-11-25 by timsifive