riscv/riscv-debug-spec #192: Why is there no register for SysBusAdd127:96 ?

mwachs5, 2018-01-03:
Don't we need a sbaddress3 register for systems that have 128-bit addresses?

timsifive, 2018-01-09:
These addresses would be physical, and I decided that 96 bits ought to be enough for everybody.

I know people might want to play games with assigning the higher bits some meaning. Do we need a full 128 bits? Reading it now I'm inclined to say yes, just to avoid future pain. We have the address space, although I'd move sbcs and sbaddress[0-2] down to make space for sbaddress3. It'd be pretty backwards-incompatible.

timsifive, 2018-01-11:
I'm inclined to fix this since we're touching the bus master anyway.

How do you want to handle versioning? The obvious thing to do is to move the existing bus address registers down one address to make space for the new one at its proper place. But then sbaddr0 lives at where sbcs used to live, so we can't version between old and new using bits in sbcs. I'm fine just incrementing the spec version at this point. Is that OK?

timsifive, 2018-01-11:
The other option is to put sbaddress3 at 0x37. It's a little awkward, but people use symbolic constants everywhere anyway, right? (Maybe it takes a few more gates in an address decoder.)

mwachs5, 2018-01-11:
Yeah, it's awkward, but we're sort of out of space for the sb* registers anyway (previously we had nice 8 round number and now we're up to 9, so it's going to be awkward somewhere).

mwachs5, 2018-01-30:
Resolved by #150

closed 2018-01-30 by mwachs5