riscv/riscv-debug-spec #194: Clarify what CSRs can be used when HARTINFO.dataccess is 0

mwachs5, 2018-01-03:
Spec says:
```
datasize: If dataaccess is 0: Number of CSR registers dedi-
cated to shadowing the data registers.

dataaddr: If dataaccess is 0: The number of the first CSR
dedicated to shadowing the data registers.
```

Does this need to clarify what registers are legal for this? RISC-V Priv spec allocates 0x7B0-0x7BF as "debug mode only CSRs", 3 of which are already defined as $dpc and $dcsr and $dscratch0 (note that $dscratch0 doesn't really need a standardized location given the current spec). Currently there is space for a system which decided it wanted to put all 12 data words in that region, but if additional functionality like #187 is actually added, it could conflict with this usage. So we may want to define a space that is "safe" to use for this.

Or should the registers used for this come out of some other space like the "non-standard R/W CSRs" region.

timsifive, 2018-01-11:
Per yesterday's meeting, Megan will discuss this with the ISA group.

mwachs5, 2018-01-12:
The "non standard R/W CSRs" can be used as well. I'll update the text to reflect this

mwachs5, 2018-01-17:
I think the whole point of the "non standard CSRs" is that they are usable for implementation-specific purposes. So stating it again seems a bit redundant.

closed 2018-01-17 by mwachs5