riscv/riscv-debug-spec #202: v13 vs master branches

mwachs5, 2018-01-08:
Why are there both `v13` and `master` branches (I believe the default working branch used to be `v13`). Should we just delete the `v13` branch? I think it is confusing at the moment. I assume this is related to #188, but not clear what the intended usage of the branches is right now.

timsifive, 2018-01-08:
`master` refers to the OpenOCD master. `riscv` is the RISC-V mainline. `v13` can be deleted.

timsifive, 2018-01-08:
I think I created `0.13` when messing with versioning. It can be deleted AFAIC.

timsifive, 2018-01-08:
github won't let me delete it because it's protected. I'm not sure how a branch gets that status.

mwachs5, 2018-01-08:
I unprotected it (via Settings->Branches). And protected `master`

timsifive, 2018-01-08:
I deleted it.

closed 2018-01-08 by timsifive

mwachs5, 2018-01-08:
hm that seems to have had the side effect of deleting all PRs being made against it

timsifive, 2018-01-08:
:-( Were there any important ones? Presumably people can recreate them if they still have them in their own git repo.

mwachs5, 2018-01-08:
They are just Closed (my 'deleted' was misleading). They can be re-opened against the `master` branch as desired. I did that for #185