riscv/riscv-debug-spec #208: Clarify trigger match logic

zhonghochen, 2018-01-12:
I have two questions about the trigger match logic:

Q1. When mcontrol.match is 1 and tdata2 is b10,
does a LW instruction to address 0 hit the trigger?

Q2. When mcontrol.match is 1 and tdata2 is b101,
does a LW instruction to address 3 hit the trigger?

It seems the trigger comparing base address without size?

--

timsifive, 2018-01-12:
On Thu, Jan 11, 2018 at 10:36 PM, zhonghochen <notifications@github.com>
wrote:

> I have two questions about the trigger match logic:
>
> Q1. When mcontrol.match is 1 and tdata2 is b10,
> does a LW instruction to address 0 hit the trigger?
>
The spec says:
"Matches when the top M bits of the value match the top M bits of tdata2. M
is XLEN-1 minus the index of the least-significant bit containing 0 in
tdata2."

So if tdata2 is b10, the index of the LSB 0 is 0, M is XLEN-1, the
comparison is that bits 31:1 equal 31'b1. Does not match 0.

> Q2. When mcontrol.match is 1 and tdata2 is b101,
> does a LW instruction to address 3 hit the trigger?
>
> If tdata2 is b101, the index of the LSB 0 is 1, M is XLEN-2, the
comparison is that bits 31:2 equal 30'b1. That does not match 3, since bits
31:2 of 3 are 0.

> It seems the trigger comparing base address without size?
>
Looks like it. I think we did have support for access size at some point,
but evidently it got removed somewhere. Is this something you need?

Tim

zhonghochen, 2018-01-13:
Current spec fulfills our usages. I just want to make sure our implementations are compatible with spec.

The first case occurs  when GDB sets a watchpoint on an halfword and the halfword is accessed by a LW instruction. For handling the case, OpenOCD needs to set a trigger on a full word, but it also introduces false matches. OpenOCD needs to parse the instruction type for checking whether the watchpoint is hit.

When misaligned load/store is supported, the trigger needs to compare a wider range.

Using mcontrol.match 2 and 3 only relieves false matches, and it takes more triggers.

It will be nice to have a new feature for improving above usage, such as adding one bit in mcontrol for comparing full accessed range of a instruction.

Zhonh-Ho

timsifive, 2018-01-15:
That's a good suggestion. Do you have any idea what such a bit should be called?

zhonghochen, 2018-01-16:
I did not notice that bit 20 is the last bit left in mcontrol for RV32.
I am hesitant to consume the last bit. So far, it's okay to let the feature pending.

timsifive, 2018-10-26:
Closing due to lack of interest/activity.

closed 2018-10-26 by timsifive