riscv/riscv-debug-spec #215: There is no way to tell which trigger fired

timsifive, 2018-02-06:
I'd like to add a R/W bit to each trigger that is set when the trigger fires, and can be cleared by the debugger. This allows debuggers to only disable the triggers that hit when stepping after hitting one, and also to provide the user more detailed information about what happened.

mwachs5, 2018-02-06:
Is there no way (by probing registers, PC, etc) to get the same information? Is this just a shortcut or completely missing functionality?

timsifive, 2018-02-06:
I'd say it's completely missing.

For example when having several address read triggers set, it may be possible to disassemble the code and read the registers to see which one was triggered (assuming it halts very close to the instruction that caused the trigger), but that's really more than you can expect a debugger to do.

mwachs5, 2018-02-06:
I'm not sure what you mean "assuming it halts very close to the instruction that caused the trigger". 

Isn't the instruction that caused the trigger reflected in $dpc?

timsifive, 2018-02-06:
The spec allows for hardware triggers to not be exact. (See \Ftiming in \Rmcontrol.)

timsifive, 2018-02-06:
Code might read:
```
sw t0, 0(a0)
j somewhere
```
Depending on implementation, the store might not cause a trigger until after the jump has been executed. At that point I don't think there's anything the debugger can do to figure out which trigger was hit.

mwachs5, 2018-02-06:
My reading of the "imprecise" timing was that if it was "before" timing it was precise (one could resume at that instruction like for all exceptions), but if it was "after" timing (like read data match) then it would break at the "end" of the instruction or on the next instruction and you shouldn't replay that instruction. 

So maybe your example is just bad, it doesn't make sense for a sw but a lw could exhibit what you mean. But it still seems like you would then break at the `j somewhere` and be able to look back 1 instruction.

No matter what, it seems like you need to have some loop over all the triggers and say "does this one apply?" by whatever mechanism you choose, so I'm not sure this proposal buys you much.

timsifive, 2018-02-06:
If \Ftiming is 1, then "[t]he action for this trigger will be taken after
the instruction that triggered it is executed. It should be taken before
the next instruction is executed, but it is better to implement triggers
and not implement that suggestion than to not implement them at all." So a
trigger might cause a halt an arbitrary number of instructions after it
actually matched.

> No matter what, it seems like you need to have some loop over all the
> triggers and say "does this one apply?" by whatever mechanism you choose,
> so I'm not sure this proposal buys you much.
>
Right now there's nothing to loop over. You can guess which trigger was
hit, but it requires assuming things about the target, possibly
disassembling instructions (what about custom extensions?), which is
painful to implement, and test. The extra bit is pretty cheap in hardware,
and completely solves all of these issues.

mwachs5, 2018-03-08:
Closed by #217

closed 2018-03-08 by mwachs5