riscv/riscv-debug-spec #233: DPC Clobbering clarification

mwachs5, 2018-03-14:
From Ben Horgan:
3.6: How does software know if the implementation clobbers the dpc when executing the program buffer?

mwachs5, 2018-03-14:
The spec already says 
```
Executing the Program Buffer may clobber \Rdpc. If that is the case, it must be
possible to read/write \Rdpc using an abstract command with \Fpostexec not set.
The debugger must attempt to save \Rdpc between halting and
executing a Program Buffer, and then restore \Rdpc before leaving Debug Mode.
```

That text was already edited to address the same question. Not sure how it can be made more clear.

timsifive, 2018-03-14:
The answer is: The debug software doesn't know, which is why it must attempt to save/restore dpc.

But I'm not sure how to add that to the spec without cluttering it up.

timsifive, 2018-05-29:
I think we should close this issue.

closed 2018-05-29 by mwachs5