riscv/riscv-debug-spec #240: Suggestion: Allow Debug Mode to be not just M mode

mwachs5, 2018-03-14:
Spec says in Debug Mode, all instructions operate as if in M mode, and MPRV is ignored. Suggestion is to add WARL bits (presumably in DCSR) to change the prv mode in which Debug Mode operates.

markuslauterbach, 2018-03-29:
Allowing the debugger to display and edit memory from different "privilege perspectives" is an often requested feature. I second that proposal.

timsifive, 2018-04-06:
@markuslauterbach by a different privilege perspective, are you interested in anything beyond the address translation? Part of me feels the address translation is well defined in the spec, and could just as easily be done by the debugger. That would be slower due to the memory reads required, but keeps the hardware simpler.

mwachs5, 2018-04-10:
Other things that might matter:
* Certain CSRs/ instructions are not accessible in different modes (for example, `mstatus` register is not accessible in anything but M-mode, so if we allow changing the priv level, does that mean we can't access `mstatus` register in Debug Mode). This is *more* restrictions than provided by the normal `MPRV`, and seems like a good way to block yourself into a corner from which you can't recover from. The omnipotent debugger can just not do those things.  So let's not make that part of the spec.

* In addition to translation there are Phyiscal memory protections and other access protections, which I think is one thing @markuslauterbach was referring to. Let's say you change the hypothetical "DPRV" to "U" and then access a memory address that the PMP settings don't allow in U-Mode. The core would get an exception, which in Debug Mode doesn't transfer the hart to the usual exception handler after updating all the `mepc`/mcause registers, it just sets the "Exception" code in `ABSTRACTCS.error`. So then debugger would have to know why it got that exception anyway.

My point is that the debugger is going to have to be pretty smart about the memory system, either way, so I tend to agree with @timsifive that the SW can and should handle this rather than adding HW that probably has a bunch of corner cases.

 Another corner case... probably we don't want anything but Debug Mode to be able to access the Debug Module/Program Buffer space. but now we have to punch a hole through for "User Mode Debug Mode" and make sure that it can it access it as well? I don't even know how that works with address translation. Does it mean that HW must enforce that debug module space is always mapped to the same place in the different memory modes?

yungchinghsiao, 2018-04-24:
We recently noticed that, in Sv32 mode, the virtual address width is 32-bit but the physical address could be as wide as 34-bit. Without some way of temporarily accessing address spaces in virtual addresses in U/S-modes, it is not possible to access the memory ranges accessible in U/S-mode.

mwachs5, 2018-04-26:
Good point, though reading the history of this feature on the mailing list it appears to be a configuration that is expected to be rarely used, so even if we add a "DPRV" to support this case we should make sure it remains optional. Perhaps a `commentary` block to explain this point to go along with the optional feature in the spec?

mwachs5, 2018-05-04:
Closed by #265

mwachs5, 2018-05-04:
BTW I did not add the suggestion about this bit getting reset every time you enter/leave debug mode. Debugger should be able to handle that just fine without putting the burden on HW.

closed 2018-05-04 by mwachs5

timsifive, 2018-05-04:
I agree.