riscv/riscv-debug-spec #242: How to enforce entering debug mode on reset

mwachs5, 2018-03-14:
Example: Debugger wants to trigger on some bad scenario, so it places  a breakpoint on the "bad scenario handler". Let's say bad scenario generally happens shortly after reset. Then core gets reset (by someone pressing reset button, or SW reset... essentially something out of debugger's control). Debugger will notice the hart got reset because of `hasreset`, but as the spec is currently written the HWBP set up to trigger on bad scenario could be lost and run past before debugger notices.

timsifive, 2018-03-15:
Simplest solution:
Add a reset-proof halt-on-reset bit to DCSR in each hart. When 1, the hart will enter debug mode immediately out of reset.

timsifive, 2018-03-23:
I'd like to get this feature in before we ratify.

I think either bit 27 in dmcontrol, or bit 3 in DCSR. Preference? It feels like it belongs more in dmcontrol so it's easier to think of it as not being reset when the hart is.

mwachs5, 2018-03-23:
If it's in `dmcontrol`, it's global. What does that mean then w.r.t. each hart?

timsifive, 2018-03-26:
Good point. If it's in dmcontrol, there would have to be set-halt-on-reset
and clear-halt-on-reset bits. DCSR seems more clean from that point of
view. Is there any issue with saying this one bit persists across reset?

timsifive, 2018-03-27:
> It's trivial to say bits aren't reset by hart reset. The harder part is
> saying that they must have SOME initial value, because now you're
> introducing the idea of two levels of reset *within the core CSRs*, which
> is a new concept that exists nowhere in RISC-V ISA land, including this
> document.

That makes me think the bits have to be in DM then, so some way of
accessing them will probably be dmcontrol. Do you agree?