riscv/riscv-debug-spec #250: Burst access to GPR/CSR without program buffer

omontfort, 2018-03-28:
If no program buffer is implemented, there is no way to perform burst accesses to the hart registers (CSR and GPR) which can have huge impact on debug performances.
Indeed the actual procedure is to perform a write access to the abstractcs register to update the regno field, then to read the data register. It requires 2 DTM accesses by register whereas a burst feature would allow to access all registers in one single DTM access.

One solution to improve this issue, could be to add a bit to enable autoincrement of regno when accessing data when autoexecdata field of the abstractauto register is set.

timsifive, 2018-03-28:
Adding an optional autoincrement bit to Access Register command seems pretty straightforward and would allow for this optimization.

I don't think it'd be terribly useful because the CSR space is relatively sparsely populated. It's probably most useful to read the GPRs which happens more often. (At least looking at OpenOCD right now, it's got all kinds of inefficiencies that would have to be addressed first and by themselves would improve performance more than this suggestion.)

jhauser-us, 2018-08-03:
While I can accept it if this feature request is rejected, I'd like to add my vote in favor of making bit 19 of the "access register" abstract command an optional `aarpostincrement`.  (The only way to use this would be with bits set in `abstractauto`.)

timsifive, 2018-08-08:
That would look something like:
```
        <field name="aarpostincrement" bits="19">
            This bit is optional.

            0: No effect.

            1: After a successful register access, \Fregno is incremented
            (wrapping around to 0).
        </field>
```

How can a debugger know whether this feature is implemented?

jhauser-us, 2018-08-10:
> How can a debugger know whether this feature is implemented?

The same as other abstract command options:  Try it, and see whether `cmderr` gets set to 2, "not supported".

timsifive, 2018-08-10:
Somehow I had thought there was an issue with that, but now I can't think of it. I should've written it down a few days ago.

closed 2018-09-07 by timsifive