riscv/riscv-debug-spec #252: SB Access: should there be an error code for unsupported SB Access size?

mwachs5, 2018-04-09:
Currently the spec calls out SB error code 3 to be "some other error, e.g. alignment". What about the error case when you try to do an unsupported sbaccess size? Does that fall under "some other error"? It seems like a common enough case that it should be specifically tied to some value, even if it is also 3.

mwachs5, 2018-04-09:
Oh I see you tried to clarify this here: https://github.com/riscv/riscv-debug-spec/pull/203/commits/b8eb62a53e9616e078d71ee264cf73528ee87ef2

But that is not really in a place that someone might look for it when implementing SBERROR logic

mwachs5, 2018-04-09:
But then it is also unclear the priority of these errors. What if I send a misaligned access to an inaccessible address, which error type should I return? Either is fine?

timsifive, 2018-04-10:
I'm updating \Fsberror to also mention unsupported access size for 3.

mwachs5, 2018-04-25:
Resolved by #254

closed 2018-04-25 by mwachs5