riscv/riscv-debug-spec #264: tdata registers shoudl be described as WARL

mwachs5, 2018-04-25:
For example, in RV64 implementing address match triggers and not data match triggers, the `tdata2` register may not actually hold a full 64-bits. They should be described as WARL.

timsifive, 2018-05-02:
WARL feels too generous. It would allow an implementation to only do address matches on 16-byte aligned addresses. That's taking things a little too far.

Furthermore, address matches are on virtual addresses, which I assume are always XLEN bits wide.

I'm all for something like "If there are bits in \Rtdatatwo that aren't useful to write, they may be tied to 0." but then you get into what is "useful."

timsifive, 2018-05-04:
Thanks for explaining the virtual addresses. I agree it makes sense to allow an implementation only to write the lower N bits of a tdata register. I'm not sure about the sign extension. A quick search through the privspec doesn't find any mention of sign extending addresses. What am I missing?

mwachs5, 2018-05-07:
For example, the Priv spec says about `mepc`: 

> 
> {\tt mepc} is a \warl\ register that must be able to hold all valid physical
> and virtual addresses.  It need not be capable of holding all possible invalid
> addresses.  Implementations may convert some invalid address patterns into
> other invalid addresses prior to writing them to {\tt mepc}.

timsifive, 2018-05-07:
I like how that's worded.