riscv/riscv-debug-spec #275: A Debug Mode trigger might be prevented by a M mode trigger

zhonghochen, 2018-05-04:
We find an issue that a debug mode trigger might be prevented by an M mode trigger.

Assuming 
   Trigger 0 is used by M Mode, and chain bit is set, and
   Trigger 1 is used by Debug Mode.

According to the spec, the M Mode trigger (trigger 0) will prevent the Debug Mode trigger (trigger 1).
https://github.com/riscv/riscv-debug-spec/blob/675bb14656f6605fcbe303de0cb16cff08301c6e/xml/hwbp_registers.xml#L169-L173

It sounds like we should add a limitation that Debug Mode should use triggers with lower indexes.

mwachs5, 2018-05-04:
Why would M-mode reserve one trigger and  set the chain bit and not reserve/set the one above it? Are you proposing a "Debugger/M-Mode SW should do this" change or "Hardware must not allow this" change?

I suppose the M-mode code may have been interrupted by the debugger in the middle of setting up the triggers and so hadn't had time to reserve the upper trigger, is that the case you are concerned about?

timsifive, 2018-05-04:
@mwachs5 The M-mode code could simply have a bug that causes it to set the chain bit.

I think the simplest solution is to have \Fchain only inhibit the next trigger from matching if both triggers have the same \Fdmode value.

zhonghochen, 2018-05-14:
The following sequence illustrates an example of the issue
Step 1. Debug-mode set a breakpoint with trigger 1
Step 2. M-mode program set chain bit of trigger 0, because it has bugs
Step 3. M-mode program executes the instruction at the breakpoint, but the breakpoint is not triggered

How about resolving the issue at implementation level while being compatible with current spec?
Hardware implementation should prevent M-mode program from setting the chain bit of a trigger, when the next trigger is debug mode.

mwachs5, 2018-05-14:
I see your point. I'm not sure which is easier for HW to implement (preventing the setting of the chain bit, or ignoring it). Is there a way we could specify either implementation?

timsifive, 2018-05-15:
In that case, hardware must implement one of:
1. Ignore \Fchain in triggers with \Fdmode=0, iff \Fdmode in the next trigger is 1.
2. Don't allow \Fchain to be set in a trigger if \Fdmode=0 in the same write, and \Fdmode of the next trigger is 1.

When setting a trigger, a debugger must ensure that the previous trigger does not have \Fchain set if \Fdmode=0.

Doesn't seem so bad, although it'd be cleaner to have a single hardware requirement.

closed 2018-05-31 by timsifive