riscv/riscv-debug-spec #276: What is meaning/rationale of the dcsr.mcause priorities?

mwachs5, 2018-05-14:
Is there a rationale behind the priorities of the different reported DCSR.mcause values? And do we need to clarify if "1" or "4" is highest priority?

timsifive, 2018-05-15:
I've never seen a question about which one is the highest priority. It's simply the one with the highest number. But I'm open to a clarification if it's confusing.

timsifive, 2018-05-17:
The implications are mostly cosmetic. IIRC the priorities was added because somebody asked what they were, and it does make sense to report the least expected cause to the user.

In the example you give (both swbp and hwbp hit, but swbp is reported), a debugger might just clear the swbp and then assume it can step off it, causing the hwbp to hit again. I think most debuggers will disable all hwbps during a single step that's intended to get off a breakpoint so in practice this is probably not a problem.

closed 2018-05-29 by timsifive