riscv/riscv-debug-spec #301: What does postexec mean when there's no Program Buffer?

timsifive, 2018-06-26:
Possibilities:
1. It's just a nop
2. The abstract command returns "not supported"

mwachs5, 2018-06-26:
I guess another (gross) question/possibility Is similar to what happens if you have only 2 words of program buffer, and you write them both with say, nop, and you have no ebreak. Program buffer execution happens but once it runs off the end... it’s undefined, right? In the implementations I am aware of you’d get an exception once you run off the end. So maybe a legitimate possibility here is getting an exception as you try to execute the “null” program buffer

timsifive, 2018-06-26:
How about saying that running a non-existant program buffer leads to undefined behavior, but we recommend it result in "not supported?"

PR #352 (2018-09-07): Clarify postexec when there is no Program Buffer
Fixes #301.
commit e0ff31ea (2018-09-07): Clarify postexec when there is no Program Buffer (#352)

Fixes #301.
closed 2018-09-07 by timsifive