riscv/riscv-debug-spec #313: Is the command register readable or write-only?

jhauser-us, 2018-07-22:
A comment in section 3.5, "Abstract Commands", says
> To discover which command failed, the debugger has to look at the state of the DM (eg. contents of `command`) or hart (eg. contents of a register modified by a Program Buffer program) to determine which one failed.

But examining `command` is possible only if the register is readable.  The section that describes the `command` register indicates that all bits of the register support only "W" access, which elsewhere is defined as:  "Write-only.  When read this field returns 0."

Since reading `command` is required to return only 0, how could a debugger look at the contents of `command` as the comment proposes?

timsifive, 2018-07-30:
That's a good point. With the current spec, the comment is clearly wrong. However, looking at `command` can be helpful to determine what has executed so far.

The reason to make `command` write-only is so implementations don't have to waste bits on storing invalid commands. We can get the best of both worlds by making `command` write-any-read-legal. How does that sound?

jhauser-us, 2018-07-31:
My vote would be to leave `command` entirely write-only and simply modify the comment.

To start with, for most examples I can think of, there's not a critical need to be able to read `command` to determine what completed and what didn't.  Yes, it could sometimes be helpful, but more often the ability to read `command` is of limited use, or no help at all.

Second, I'm not certain WARL is always well-defined for unsupported commands.  Would the value retained in `command` really be consistent enough across all DM implementations that debuggers could reliably interpret what they see from reading it?  I can imagine this working for certain tested combinations of debuggers and targets, but not generally.  

Furthermore, for implementations that choose not to support `abstractauto`, there's currently no requirement AFAIK to store the bits written to `command`.  A command could be executed immediately and the command bits discarded.  Making `command` WARL would require the extra flip-flops, for (in my opinion) minimal benefit.

You could split the difference and make it optional whether `command` can be read back; otherwise, you read zeros, as currently.  However, by shrinking the mandate, I think you'd reduce the chance that a generic debugger would bother to try, since it must have a fall-back coded anyway.  The space of DM implementations is already greatly fragmented, placing a lot of burden on debuggers to adapt dynamically to what is and isn't supported by a DM.  I can picture debugger authors, already exhausted by all the other options, not bothering to take advantage of the possibility of reading back `command` on the slim chance that it's supported.

All that being said, it won't break my heart if this is decided a different way.

PR #328 (2018-08-10): \Rcommand is not readable.
Fixes #313.
commit 57df3f3b (2018-08-10): \Rcommand is not readable. (#328)

Fixes #313.
closed 2018-08-10 by timsifive