riscv/riscv-debug-spec #314: Are all program buffer implementations required to support jumps out of the program buffer?

jhauser-us, 2018-07-22:
Section 3.6, "Program Buffer", says
> The debugger can write whatever program it likes (including jumps out of the Program Buffer) ...

Later we have in a comment:
> The slightly inconsistent behavior with a Program Buffer of size 1 is to accommodate hardware designs that prefer to stuff instructions directly into the pipeline when halted, instead of having the Program Buffer exist in the address space somewhere.

If a DM implements the program buffer by injecting instructions directly into the pipeline of a halted hart, what is supposed to happen when the instruction being injected is a branch that jumps out of the program buffer?  The first quotation above suggests that such jumps must be supported, yet  it will be costly for a DM of this type to do so.

(Please be aware that a DM might prefer to implement program buffer execution by instruction injection even when the program buffer is larger than one word.)

timsifive, 2018-07-30:
There's definitely a conflict here. On the one hand, jumps allow for a lot of flexibility, especially when it comes to quickly letting a debugger perform complex operations. On the other hand, we'd like to support instruction stuffing implementations.

A general debugger doesn't need to use any jumps (eg. OpenOCD doesn't use any) so they'd be limited to special cases. In order to preserve the possibility, but not mandate it, I think the best solution is a bit in dcsr that indicates whether the hart allows jumps/branches in Program Buffer programs.

How does that sound?

jhauser-us, 2018-07-31:
There are potentially two overlapping issues here:  First, are *any* branches supported in the program buffer?  (Can the program buffer contain a loop, for example?)  Second, if the answer to the first question is "yes", then is it possible to branch to an address outside the program buffer?  (Can the program buffer call a subroutine that exists in regular memory, for example?)

I expect, like OpenOCD, debuggers can normally manage well enough without any branches in the program buffer.  On the other hand, there are a few tasks that might benefit from program buffer loops, such as zeroing or copying a block of memory.  Even an instruction-stuffing DM, at additional cost, could support program buffer loops without supporting branches outside the program buffer.

So perhaps two bits in `dcsr` are justified, with this combination of values:
  00 = Branches are not supported at all in the program buffer.
  10 = Branches are supported in the program buffer, so long as the branch target is also in the program buffer.
  11 = All branches are supported in the program buffer, even to target addresses outside the program buffer.

The alternative is to mandate that some of these three options aren't allowed.  If you want to restrict to a single case, I would argue it should be the first one (no branches) and not the third one (all branches).  Alternatively, one might exclude only the middle case, which is what it sounds like you were prepared to do.

A releated issue, once branches may be forbidden, is to specify precisely what happens if they're encountered.  I would recommend that they be treated the same as illegal instructions, thus causing `cmderr` to be set to 3 ("exception"), not 2 ("not supported").  In other words, for some RISC-V cores, some or all branch instructions would be illegal when executing in debug mode.

timsifive, 2018-08-02:
Agreed. Except I flipped the bits for backwards compatibility.

jhauser-us, 2018-08-03:
As a mostly aesthetic tweak, any chance this new field could be shifted over closer to the `xdebugver` field, such as bits 27:26?  They both provide constant information to the debugger, as opposed to the other fields in `dcsr` which are all active in some way.

closed 2018-09-07 by timsifive