riscv/riscv-debug-spec #316: Are any dmcontrol actions allowed during an abstract command?

jhauser-us, 2018-07-24:
When an abstract command is executing (`abstractcs.busy` is set), what happens if a `dmcontrol` action (`haltreq`, `resumereq`, `hartreset`, `ackhavereset`, `setresethaltreq`, `clrresethaltreq`) is initiated for the same hart?

timsifive, 2018-07-31:
Don't use haltreq/resumereq while executing an abstract command: #324 
ack* seems like it should be possible
resetting should also be possible. If an abstract command hangs you want to still be able to reset the system.

allenjbaum, 2018-08-22:
As an aside: are there any legitimate use cases for writing dmcontrol with more than one of the haltreq/resumereq/hartreset/ackhavereset/setsresethaltreq/clrresethaltreq bits set?
I wouldn't object if the spec said "don't do that" so we don't have to validate what happens if they try...

timsifive, 2018-08-24:
I could make an argument that a debugger might set resumereq, ackhavereset, and clrresethaltreq all together (eg. when halted out of reset, but the following reset it doesn't want to be halted). But it's not hard for a debugger not to do that. It's a bit awkward with the encoding we've chosen, though. If other people think it's a good idea to add more limits, I'm happy to write up a PR.

allenjbaum, 2018-08-24:
Requiring only 1 out of those 6 bits be set is not terribly challenging if the obvious solution is to just write them serially. It does greatly reduce the state transitions, validation effort, and probably simplifies the spec itself to make multiple bits set a prohibition. I'm not sure why I left out hartreset from that list, actually.

closed 2018-09-07 by timsifive

allenjbaum, 2018-09-07:
Wasn't this already settled?
I'd prefer that at most one action be specified at a time, regardless of
whether an abstract command is in progress or not.
As you say, it easy for the debugger to not do that (just send one at a
time). This avoids validation issues, compliance issues, and enables an
implementation to save a bunch of global wires, since the command can be
encoded into 3 bits instead of 6 or 7. The downside is that in uncommon
cases (like the example above) it takes 2 additional writes - which seems
like a small price to pay, frankly.