riscv/riscv-debug-spec #318: Does accessing a CSR using an abstract command cause the usual side effects?

jhauser-us, 2018-07-24:
For any particular RISC-V CSR, reading/writing the register may be defined to have side effects.  When an implementation supports accessing a hart's CSR using a DM abstract command, is it required that such an access does, or does not, trigger all defined side effects?

For some CSRs, there might be a debugging advantage to "peeking" at its value without causing the side effects.  On the other hand, this ability could be considered incongruent with normal CSR reads.

Implementors are already free to choose not to support abstract command accesses for any individual CSRs when the implementation would be too problematic or costly.  The question is about standardizing the behavior when the choice is made to support such accesses.

One possible resolution would be to require:
- if reading a CSR normally has side effects, those side effects are not triggered when the register is read using an abstract command; and
- if writing a CSR normally has side effects, writes to the register using an abstract command must not be supported (cmderr = 2, "not supported").

Such a choice would make it safe for a debugger to blithely attempt to examine all CSRs of any hart without fear of triggering unintended consequences.

timsifive, 2018-08-02:
I always assumed that side effects do occur. The best solution is probably to add a bit to the abstract register access command that inhibits side effects. Implementations may support with/without/both.

mwachs5, 2018-08-02:
What CSRs have side effects, other than the side effect of changing their
own state when written?

timsifive, 2018-08-02:
The ISA spec explicitly states that CSR writes may have side effects:
> The CSRRS (Atomic Read and Set Bits in CSR) instruction reads the value of the CSR, zero-
> extends the value to XLEN bits, and writes it to integer register rd. The initial value in integer
> register rs1 is treated as a bit mask that specifies bit positions to be set in the CSR. Any bit that
> is high in rs1 will cause the corresponding bit to be set in the CSR, if that CSR bit is writable.
> Other bits in the CSR are unaffected (though *CSRs might have side effects when written*).

I'm not aware of any standard CSRs that actually do have side effects when written.

jhauser-us, 2018-08-03:
Side effects are considered possible both on reading and writing CSRs.  Some additional quotes from the RISC-V ISA spec:
> If rd=x0, then the instruction shall not read the CSR and shall not cause any of the side-effects that might occur on a CSR read.

> Note that if rs1 specifies a register holding a zero value other than x0, the instruction will still attempt to write the unmodified value back to the CSR and will cause any attendant side effects.

> For CSRRWI, if rd=x0, then the instruction shall not read the CSR and shall not cause any of the side-effects that might occur on a CSR read.

jhauser-us, 2018-08-03:
> The best solution is probably to add a bit to the abstract register access command that inhibits side effects. Implementations may support with/without/both.

That solution obviously covers all bases.  The only possible objection is that it adds yet more fragmentation to DM implementations.  I'm not necessarily objecting myself, but I wonder if it's possible to get more community input on this question.

timsifive, 2018-08-07:
I'll bring this up during tomorrow's call. Feel free to e-mail the Kavi group as well for a wider audience.

timsifive, 2018-08-08:
On the call the consensus was that CSR accesses must always have side effects. If you don't want side effects, it was suggested to add a shadow CSR that provides access to the same data without a side effect.

I don't see you on the RISC-V Debug Group roster. Are you a Foundation Member? If so, you could join the mailing list and discuss further there.

timsifive, 2018-08-10:
Agreed.

closed 2018-08-18 by jhauser-us