riscv/riscv-debug-spec #336: Run Control clarification - running signal

bdwyatt, 2018-08-16:
The Run Control section describes when the running signal should be asserted, but never explicitly states when it would be deasserted. My assumption would be that running should be deasserted when the hart responds to a halt request.

I took a look at the Spike source, and noticed that running is treated as simply being the inverse of halted, which leads me to another point. It's also unclear if there's a valid case where both running and halted would be deasserted at the same time, and what that would represent. I would guess that it is a valid case, and would be a way to determine if a hart is unavailable.

timsifive, 2018-08-16:
Eg. the definition for allrunning says: "This field is 1 when all currently selected harts are running." Is it necessary to state that "This field is 0 when one or more currently selected harts are not running."?

The availability of a hart can be determined by looking at the *unavail bits. I would have thought that *running and *halted don't contain any useful information in that case.

What clarification would you like to see?

bdwyatt, 2018-08-16:
Tim, I'm referring specifically to section 3.4, and the signals coming from the hart to let Debug Module set the dmstatus bits.

[image]

timsifive, 2018-08-16:
I see. That is a bit odd to not mention running in the 2nd paragraph, while it is mentioned in the 3rd paragraph. The PR addresses that.

closed 2018-08-21 by bdwyatt