riscv/riscv-debug-spec #346: System reset

zarubaf, 2018-08-30:
The `dmactive` says: When `1` no other mechanism should exist that may result in resetting the Debug Module after power up. 

That may be obvious to others but do you suggest that also an external (hard) reset pin can't reset the system or are you only specifying that the `ndmreset` register can not reset the DM + DTM?

zarubaf, 2018-08-30:
I might need to add that I've implemented the first option (assuming there is a PoR circuit) but its quite convoluted and I don't like it too much. 

P.S.: Thanks for all your work

timsifive, 2018-08-31:
I feel that if there's a real reset pin, then it's OK to reset absolutely everything. Perhaps @mwachs5 has more insight here. She's put much more thought into reset than I have.

mwachs5, 2018-08-31:
I'd say that if you are interested in allowing users to debug across an external system reset, then that external system reset can't reset DM + DTM. But it is definately harder to provide that capability as you do need some OTHER reset (as you said, like a POR) to reset the DM and DTM.

timsifive, 2018-09-07:
The language that specifically includes the system reset as not resetting the DM, was added in c08486f08fcbb3f9e0542dad83670df6f043be4d. I'm inclined to change that to only suggest that external global reset excludes the DM, because reset can get quite complex and is highly implementation-dependent.

PR #350 (2018-09-24): Allow global reset to reset the DM.
Fixes #346.
commit 10dfa65c (2018-09-24): Allow global reset to reset the DM. (#350)

Fixes #346.
closed 2018-09-24 by timsifive